rusefi/rusefi #7878: JAXBException is back

rusefillc, 2025-05-13:
console does not work right after update

work-around: run console again

rusefillc, 2025-05-13:
with uaEFI I have

```
Performing action on port COM53...
Received a signature rusEFI master.2024.11.17.uaefi.2834573262
Page size is 22960
Reading current calibrations...
Backing up current ini-file `C:\Users\superuser\.rusEFI\ini_database\2834573262.ini`...
`2834573262.ini` ini-file is backed up as `state/2025-07-13-10.07.34_prev_calibrations.ini`
Backing up calibrations to files `state/2025-07-13-10.07.34_prev_calibrations.zip` and `state/2025-07-13-10.07.34_prev_calibrations.msq`...
Backing up current calibrations failed: javax.xml.bind.JAXBException
 - with linked exception:
[java.util.zip.ZipException: ZipFile invalid LOC header (bad signature)]
Resuming port scanning...
Port scanning is resumed.
Failed to back up current calibrations...
hint: error state is already in your clipboard, please use PASTE or Ctrl-V while reporting issues
```

rusefillc, 2025-05-16:
https://github.com/rusefi/rusefi/issues/5956

kifir23917, 2025-05-16:
```
E 250516 164404.710 [ECU Communication Executor1] CalibrationsHelper - Backing up calibrations failed:
javax.xml.bind.JAXBException
 - with linked exception:
[java.util.zip.ZipException: ZipFile invalid LOC header (bad signature)]
	at javax.xml.bind.ContextFinder.firstByServiceLoaderDeprecated(ContextFinder.java:590)
	at javax.xml.bind.ContextFinder.find(ContextFinder.java:362)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:691)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:632)
	at com.rusefi.xml.XmlUtil.writeXml(XmlUtil.java:24)
	at com.rusefi.tune.xml.Msq.writeXmlFile(Msq.java:85)
	at com.rusefi.binaryprotocol.BinaryProtocol.saveConfigurationImageToFiles(BinaryProtocol.java:376)
	at com.rusefi.maintenance.CalibrationsHelper.backUpCalibrationsInfo(CalibrationsHelper.java:165)
	at com.rusefi.maintenance.CalibrationsHelper.lambda$readAndBackupCurrentCalibrations$1(CalibrationsHelper.java:213)
	at com.rusefi.maintenance.BinaryProtocolExecutor.lambda$execute$0(BinaryProtocolExecutor.java:36)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:842)
Caused by: java.util.zip.ZipException: ZipFile invalid LOC header (bad signature)
	at java.base/java.util.zip.ZipFile$ZipFileInputStream.initDataOffset(ZipFile.java:929)
	at java.base/java.util.zip.ZipFile$ZipFileInputStream.read(ZipFile.java:940)
	at java.base/java.util.zip.ZipFile$ZipFileInflaterInputStream.fill(ZipFile.java:457)
	at java.base/java.util.zip.InflaterInputStream.read(InflaterInputStream.java:158)
	at java.base/java.io.FilterInputStream.read(FilterInputStream.java:132)
	at java.base/sun.nio.cs.StreamDecoder.readBytes(StreamDecoder.java:270)
	at java.base/sun.nio.cs.StreamDecoder.implRead(StreamDecoder.java:313)
	at java.base/sun.nio.cs.StreamDecoder.read(StreamDecoder.java:188)
	at java.base/java.io.InputStreamReader.read(InputStreamReader.java:177)
	at java.base/java.io.BufferedReader.fill(BufferedReader.java:162)
	at java.base/java.io.BufferedReader.readLine(BufferedReader.java:329)
	at java.base/java.io.BufferedReader.readLine(BufferedReader.java:396)
	at javax.xml.bind.ContextFinder.firstByServiceLoaderDeprecated(ContextFinder.java:578)
	... 12 more

```

kifir23917, 2025-05-19:
Let's try to run `rusefi.console.exe` in a new process, not in `rusefi_autoupdate.exe` process to avoid problems with class loaders: https://github.com/rusefi/rusefi/pull/7915

rusefillc, 2025-05-19:
@kifir23917 closing?

closed 2025-05-19 by rusefillc