rusefi/rusefi #7897: looks like window sampling single/multiple cyl is broken differently

rusefillc, 2025-05-15:
looks like window sampling single/multiple cyl is broken differently - we are unable to move start angle, it seems to be locked at zero regardless of setting

@FDSoftware please https://github.com/rusefi/rusefi/wiki/TDB-Test-Driven-Bugfixing

* please ADD new test with 60/2 trigger in addition to current test which uses plan trigger
* in that new test please use start angle of 200

rusefillc, 2025-05-15:
@FDSoftware see suggested next step https://github.com/rusefi/rusefi/pull/7900

rusefillc, 2025-05-25:
@FDSoftware is there any connection with https://github.com/rusefi/rusefi/issues/5894?

FDSoftware, 2025-05-26:
> [@FDSoftware](https://github.com/FDSoftware) is there any connection with [#5894](https://github.com/rusefi/rusefi/issues/5894)?

yes, my next pr will be one closing the gap on that issue, but without modifying the injection/ignition code for now [so that issue would have to remain open for a bit after], just adding `getAngleOffset`

FDSoftware, 2025-06-06:
> [@FDSoftware](https://github.com/FDSoftware) are we good to close?

yup!, we also need to close/reject this pr:
https://github.com/rusefi/rusefi/pull/7900 since all the changes are in master

closed 2025-06-06 by rusefillc