rusefi/rusefi #7913: 40% of Windows 11 do not access rusEFI MSD

rusefillc, 2025-05-18:
one ECU with nice SD card, one USB cable
five Windows 11 laptops (x4 Dell and one IBM)

three laptops would access .ini file drive and SD card MSD

two laptops would not access .ini drive and SD card via MSD

rusefillc, 2025-05-18:
[image]

rusefillc, 2025-05-18:
[image]

ggurov, 2025-05-18:
is this pass or fail ? 

uaefi, master from build_server

[image]

[image]

[image]

[image]

only F:\ is accessible

[image]

[image]

[image]

rusefillc, 2025-05-18:
@ggurov pass is having file explorer showing .ini in one folder, and "new folder" test folder on SD card

still not sure how file expolorer correlates with all internal diagnostics :(