rusefi/rusefi #7952: MAF Filtering

Robb235a, 2025-05-24:
**Is your feature request related to a problem? Please describe.**
Running uaEFI on MAF Air Charge causes unsteady engine idling and operation. Engine runs fine on speed density. MAF translation table is correct. 

**Would we need new configuration fields?**
Yes, I believe a MAF filtering option would be helpful. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

An option to filter and smooth the MAF signal to facilitate smoother engine operation. 

**Describe alternatives you've considered**
Perhaps an RC low pass filter? But guessing at what resistor and capacitor would be tedious. A software solution makes more sense. 

**Additional context**
https://media.discordapp.net/attachments/1113308614195695676/1374873396004913202/image.png?ex=6832ede1&is=68319c61&hm=016ad99a9406efe53d5bfd63de08e248a8a7fa5d870545d22b0de98f23f87644&

Log: https://drive.google.com/file/d/1wiHaerLKcfCcih4x2WnUJqCeWRYE12HT/view?usp=drivesdk

[image]
[image]

rusefillc, 2025-05-24:
> Running uaEFI on MAF Air Charge 

please spell out exact input pin since i assume uaEFI has different RC on different inputs?

We can/shall also spell out RC filters on the pinout

ElDominio, 2025-05-25:
Is this  not a frequency based MAF being used on an analog input? What MAF is this that needs analog voltage smoothing?

Robb235a, 2025-05-25:
> Is this not a frequency based MAF being used on an analog input? What MAF is this that needs analog voltage smoothing?

This is a Toyota MAF that outputs a 0-5v signal. Not frequency like I think some GM MAFs are. 

[image]

rusefillc, 2025-08-24:
> Maybe someone can implement filtering code from FOME file to work in rusefi https://github.com/FOME-Tech/fome-fw/blob/master/firmware/init/sensor/init_maf.cpp

Done

closed 2025-08-24 by rusefillc