rusefi/rusefi #7962: rusEFI console crashes after 60 seconds

dron0gus, 2025-05-27:
Local build.
Linux host.
MRE and UAEFI same result.

Full log also attached.

[console_crash.txt](https://github.com/user-attachments/files/20458417/console_crash.txt)

```
I 250527 164255.626 [SerialPortScanner inspectPort2] SerialIoStream - ttyS4: Closing port...
I 250527 164255.686 [ECU Communication Executor1] BinaryProtocol - Got livedata
I 250527 164255.695 [SerialPortScanner inspectPort2] SerialIoStream - ttyS4: Closed port.
I 250527 164255.695 [SerialPortScanner inspectPort2] SerialPortScanner - Port ttyS4 does not look like an ECU
I 250527 164255.738 [ECU Communication Executor1] IncomingDataBuffer - load image offset=0 header: timeout 5000ms. Got only 0 byte(s) while expecting 2
I 250527 164255.738 [ECU Communication Executor1] BinaryProtocol - readImage: ERROR UNEXPECTED Something is wrong, retrying... NO RESPONSE
W 250527 164255.743 [ECU Communication Executor1] ConfigurationImageFile - No image to save
W 250527 164255.743 [ECU Communication Executor1] BinaryProtocol - No image for saveConfigurationImageToFiles
E 250527 164255.743 [ECU Communication Executor1] IoUtils - CONNECTION FAILED, did you specify the right port name?
java.lang.Exception
	at com.rusefi.util.IoUtils.exit(IoUtils.java:16)
	at com.rusefi.io.LinkManager$3.onConnectionFailed(LinkManager.java:102)
	at com.rusefi.io.serial.PortHolder.connectAndReadConfiguration(PortHolder.java:64)
	at com.rusefi.io.serial.StreamConnector.lambda$connectAndReadConfiguration$0(StreamConnector.java:39)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:840)
I 250527 164255.744 [Thread-0] PersistentConfiguration - Shutdown hook...
```

dron0gus, 2025-06-01:
Now console crashes without 60sec delay. 

[console_crash_2.txt](https://github.com/user-attachments/files/20540384/console_crash_2.txt)

```
I 250601 163253.705 [ECU Communication Executor1] BinaryProtocol - Reading from controller rusEFI master.2025.06.01.alphax-4k-gdi.633603187
I 250601 163253.793 [SerialPortScanner inspectPort1] SerialIoStream - ttyACM0: Closing port...
I 250601 163253.830 [SerialPortScanner inspectPort1] SerialIoStream - ttyACM0: Closed port.
I 250601 163253.830 [SerialPortScanner inspectPort1] SerialPortScanner - Port ttyACM0 looks like an ECU
I 250601 163253.830 [SerialPortScanner inspectPort1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort ttyACM0
I 250601 163253.832 [SerialPortScanner inspectPort1] SerialIoStream - ttyACM0: Closing port...
I 250601 163253.873 [ECU text pull1] BinaryProtocol - Port shutdown: Stopping text pull
I 250601 163254.346 [SerialPortScanner inspectPort1] SerialIoStream - ttyACM0: Closed port.
I 250601 163254.347 [SerialPortScanner inspectPort1] SerialPortScanner - ECU at ttyACM0 does not have an OpenBLT bootloader
I 250601 163255.738 [Ports Scanner] SerialPortScanner - Port ttyS4 timed out, adding as Unknown.
I 250601 163255.739 [Ports Scanner] SerialPortScanner - Port ttyACM0 detected as: ECU
I 250601 163255.745 [Ports Scanner] SerialPortScanner - Port ttyS4 detected as: Unknown
I 250601 163255.755 [AWT-EventQueue-0] Launcher - Rendering available ports: [ttyACM0 (ECU), ttyS4 (Unknown), SocketCAN (CAN)]
I 250601 163256.059 [Ports Scanner] TcpConnector - Connection refused in getAvailablePorts(): simulator not running in 1ms
I 250601 163256.273 [ECU Communication Executor1] IncomingDataBuffer - [hello] header: timeout 5000ms. Got only 0 byte(s) while expecting 2
I 250601 163256.274 [ECU Communication Executor1] BinaryProtocol - No signature returned by serial=ttyS4
E 250601 163256.274 [ECU Communication Executor1] IoUtils - CONNECTION FAILED, did you specify the right port name?
java.lang.Exception
	at com.rusefi.util.IoUtils.exit(IoUtils.java:16)
	at com.rusefi.io.LinkManager$3.onConnectionFailed(LinkManager.java:102)
	at com.rusefi.io.serial.PortHolder.connectAndReadConfiguration(PortHolder.java:64)
	at com.rusefi.io.serial.StreamConnector.lambda$connectAndReadConfiguration$0(StreamConnector.java:39)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:840)
I 250601 163256.275 [Thread-0] PersistentConfiguration - Shutdown hook...
```

rusefillc, 2025-06-09:
@dron0gus issue or not?