rusefi/rusefi #7979: TS Ultra Tuning & Dyno views

rusefillc, 2025-05-30:
TS Ultra has extra views which do not work with RE

[image]

rusefillc, 2025-05-30:
@ggurov any chance you can help/tell what needs to be done?

ggurov, 2025-05-30:
they are shipped with tunerstudio, they can be created and saved, but can't put them into the ini. i can provide some samples for stuff i tried to shoehorn in following speeduino examples later, the views are very powerful in a way that they can combine table editor, dialogs, gauges, entry boxes, etc on the same page.

rusefillc, 2025-05-30:
@ggurov yes I would be interested!

ggurov, 2025-06-20:
> [@ggurov](https://github.com/ggurov) how can we support/be compatible with stock tune views? Having same names for tables/live data is enough?

i think the actual TS installation controls the enable flag, and it checks the signature for what to load