rusefi/rusefi #8013: Let's go over all tables again make sure neither are square

rusefillc, 2025-06-04:
Let's double-check that all tables follow #8004 convention to reduce chance of #6944 issue

FDSoftware, 2025-09-21:
also to fix on the same scope:

https://github.com/rusefi/rusefi/issues/8609
https://github.com/rusefi/rusefi/issues/8610

rusefillc, 2025-10-28:
``! asserting that rectangular configuration works``

```
	int16_t[CLT_TIMING_CURVE_SIZE x CLT_TIMING_CURVE_SIZE] autoscale ignitionCltCorrTable;;"deg", 0.1, 0, -50, 50, 1
    int16_t[CLT_TIMING_CURVE_SIZE] autoscale ignitionCltCorrTempBins;CLT-based timing correction;"SPECIAL_CASE_TEMPERATURE", 1, 0, -40, @@CLT_UPPER_LIMIT@@, 0
    uint8_t[CLT_TIMING_CURVE_SIZE] autoscale ignitionCltCorrLoadBins;;"Load", 5, 0, 0,  @@MAP_UPPER_LIMIT@@, 0
```