rusefi/rusefi #8056: Haltech 0x361 CANBus message - Engine demand

ric355, 2025-06-10:
https://github.com/rusefi/rusefi/blob/a7f300479bfe857c69a1bec6dde1562b1e10eec6/firmware/controllers/can/can_dash_haltech.cpp#L38

I was looking at the Haltech specification today. I see that Engine Demand is a percentage, with a factor of 10 (it is defined as y = x / 10), and that you are using Manifold Pressure as a proxy for engine load.

I'm assuming it was decided that load can go above 100% on boosted vehicles, and therefore it is already considered a percentage. However, I think this value should be multiplied by 10 before sending, based on what the Haltech specification says.

rusefillc, 2025-06-10:
https://xyproblem.info/

https://en.wikipedia.org/wiki/BLUF_(communication)

ric355, 2025-06-10:
I won't bother any more then.

closed 2025-06-10 by rusefillc