rusefi/rusefi #8110: MLV and VEAL need engine state

rusefillc, 2025-06-19:
[image]

@ggurov says

```
uint8_t Engine;Engine;"", 1, 0, 0, 0, 0

static int packTSEngineMode() {
    return      ((engine->rpmCalculator.isRunning() == 1) << 0 |                                              // IS_RUNNING
             ((engine->rpmCalculator.isCranking() == 1) << 1) |                                           // IS_CRANKING
             ((engine->fuelComputer.running.postCrankingFuelCorrection != 1) << 2) |                      // ASE
             ((engine->fuelComputer.running.coolantTemperatureCoefficient != 1) << 3) |                   // WUE
             (engine->module<TpsAccelEnrichment>()->extraFuel > 0 ||                                    // extra fuel from TPS ae decay
              engine->module<TpsAccelEnrichment>()->isAboveAccelThreshold                               // above tps threshold, this one is too fast . accel active TPS only for now (+map prediction) TODO: ask wall fuel if it's active maybe ?
              ) << 4 |
             ( (engine->module<TpsAccelEnrichment>()->isBelowDecelThreshold ||                            // decel tps ae doesn't really take fuel away TODO: ask wall fuel if it's taking fuel away ?
               engine->module<DfcoController>()->cutFuel() ) << 5) |                                      // DFCO
                 (0 << 6) | //  no map accel mode
                (0 << 7)); // no map decel mode

}

tsOutputChannels->engineMode = packEngineMode();
    tsOutputChannels->Engine = packTSEngineMode();

```

right around there
that should prevent MLV and VEAL from making holes in people's maps
cause it sets TPS DECEL on DFCO

ggurov, 2025-06-19:
false at the end means always on and not adjustable by user

```
[VeAnalyze]

	filter = std_DeadLambda ; Auto build

	filter = decelFilterMap,"After Start Enrich", engine,			&	, 4 ,		, false
	filter = accelFilterTp, "TP Accel Enrich"	, engine,			&	, 16,		, false
	filter = decelFilterTp, "TP Decel Enrich"	, engine,			&	, 32,		, false
	filter = accelFilterMap,"MAP Accel Enrich"  , engine,			&	, 64,		, false
	filter = decelFilterMap,"MAP Decel Enrich"  , engine,			&	, 128,		, false
	; this works, just another way to skin a cat
	;filter = decelFilter, "Decel Flag" , tpsaccden,			>		, 0,		, false
	filter = overrunFilter, "Overrun"	, actualLastInjection,  =		, 0,				, false

```

ggurov, 2025-06-29:
one thing to note, the 

uint8_t Engine;Engine;"", 1, 0, 0, 0, 0

capitaliation needs to be like that, it looks like logged column name needs to be "Engine", actual channel name can be lower case.

closed 2026-09-16 by rusefillc