rusefi/rusefi #8198: usability: a way to see that ignition table is not used if idling and "use separate table"

rusefillc, 2025-07-04:
we have "use idle table' set to yes and we idle

problem: ignition table looks like it's active

possible solution: remove the dot? copy "use separate table" setting to ignition table?

rusefillc, 2025-07-04:
@FDSoftware let's try smart dot approach?

meaning let's go from 
```
		xBins		= ignitionRpmBins,  RPMValue
		yBins		= ignitionLoadBins,  ignitionLoad
```

to
```
		xBins		= ignitionRpmBins,  rpmForIgnitionTableDot
		yBins		= ignitionLoadBins,  loadForIgnitionTableDot
```
in order to not have dot while we take timing from the other source?

FDSoftware, 2025-07-06:
testing with:

```c++
	if(engineConfiguration->useSeparateAdvanceForIdle && engine->module<IdleController>()->isIdlingOrTaper()){
		engine->ignitionState.rpmForIgnitionIdleTableDot = rpm;
		engine->ignitionState.rpmForIgnitionTableDot = NAN;
		engine->ignitionState.loadForIgnitionTableDot = NAN;
	} else {
		engine->ignitionState.rpmForIgnitionIdleTableDot = NAN;
		engine->ignitionState.rpmForIgnitionTableDot = rpm;
		engine->ignitionState.loadForIgnitionTableDot = engineLoad;
	}
``` 

not idling, note no green dot on the idle adv table
[image]

now idling, we have a green dot at the correct place, but the idle table dot skyrockets to the last value of the table :c

[image]

same test using 0 as "no dot"

not idle:

[image]

idle:

[image]

same test but using -1:

[image]

[image]

I assume the best scenario for now is going to be to use -1, the idle table still looks (somewhat) weird, but it's better than using NAN

FDSoftware, 2025-07-06:
@rusefillc 
 NAN: 2D no dot, 3D table broke on the right side (Then when it has a valid value again the indicator is still broken)
-1 or 0: 2D table & 3D table on weird position (but on the left side, like if we are under the table rpm bins)

also maybe we need a indicator?, this looks nice i think
[image]

[image]

FDSoftware, 2025-07-07:
@rusefillc  Is there anything else to do with this issue after #8213 or can we close it now?