rusefi/rusefi #8202: Map3D_t vs interpolate3D

rusefillc, 2025-07-04:
Any reason for ``tps_tps_Map3D_t tpsTpsMap`` to use Map3D with initTable? Let's go to simpler (?) interpolation3d approach?

Maybe go over some other usages of initTable approach without any merits?

FDSoftware, 2025-07-10:
@rusefillc:

ve_Map3D_t / pedal2tps_t / mapEstimate_Map3D_t: They have to stay as map3d because we are currently using them in the tests to mock table values.

gppwm_Map3D_t needs a bit of refactor but can use interpolation3d

script1 to script4 tables also needs refactor but can use interpolation3d

Of the two things that need refactoring, Lua's are the ones that require the least code modification. Should I continue with those?

rusefillc, 2025-07-10:
Did we make a mistake and made code less testable? Do we better stop and document that benefit of table class?