rusefi/rusefi #8260: Main relay fully broken

Nikolai521, 2025-07-10:
**Describe the bug**
On the last stable main relay works ok, but after some bundles, including the last one (Nightly 2025-07-10) the main relay function does not work at all, including the "test relay" button is dead, i tried to assign the output to another relay/solenoids and none of them works as "main relay", the output is just dead, including the test

**Hardware**
 - Board: UAEFI rev D
 - Engine info TU3JP

**To Reproduce**
Update to Nightly 

**Expected behavior**
Working main relay

**Screenshots**

[image]

**Additional context**

I tried 2 previous builds and reflashing from zero and still the same, no more context for now

rusefillc, 2025-07-10:
@Nikolai521 please help by being more specific

https://github.com/rusefi/rusefi/releases gives you all daily builds

which one is the last which works and which one is first which does not work?

Nikolai521, 2025-07-10:
> which one is the last which works and which one is first which does not work?

This was strange, just flashed the 2025-07-09 firmware and got a error uploading, retried and now the relay works, tried the day 10 bundle (last one that failed) and the relay works as expected, looks like it just fixed itself somehow

rusefillc, 2025-07-10:
@Nikolai521 sounds like the issue was about something else

closed 2025-07-10 by rusefillc