rusefi/rusefi #8273: random PPS dropouts to 0 on rawPpsPrimary, but not on rawRawPpsPrimary causing ETB/fueling drama

ggurov, 2025-07-11:
**Describe the bug**
rawPps and rawTps (and consequently TPS and Pedal Positions) drop to 0 while engine running, causing a ripple through the rest of the operation.

**Hardware**
 - Board: some f407 discovery ecu

**To Reproduce**
Reproduction steps unclear at this point yet .evidence exists in the SD card logs, which are fast enough to capture this. tunerstudio logs are less able to capture the actual dropouts. seemingly leaving a board alone with sd card logging is said to record these events.

**Expected behavior**
Tps and pps not dropping out unexpectedly.

**Screenshots**
real engine at 5200 rpm, accelerating sees a pedal and tps channels drop out for 40 ms and come back up.

when it comes back TPS AE fuel is added
and DC: output0 is pegged to max, presumably to catch the position
a ripple in batV can be seen as DC driver draws more current

[image]

thing to note here  is 3rd lane yellow and green lines

yellow line is rawRawPpsPrimary this does not drop out

which comes from 

```
	void updateUnfilteredRawValues() {
	  engine->outputChannels.rawRawPpsPrimary = m_pri.adc == nullptr ? 0 : m_pri.adc->sensorVolts;
	  engine->outputChannels.rawRawPpsSecondary = m_sec.adc == nullptr ? 0 : m_sec.adc->sensorVolts;
	}

```

green line, which is rawPpsPrimary which comes from status_loop

```
	engine->outputChannels.rawPpsPrimary = Sensor::getRaw(SensorType::AcceleratorPedalPrimary);
```

does drop out to 0, indicating sensor disappears for some amount of time.

Orange line in the same lane, shows added fuel from TPS AE. this is at 5200 rpm , full trottle for at least 2 seconds, quite steady state. 

[image]

rusefillc, 2025-07-11:
@ggurov what firmware - before or after Chibi migration?

ggurov, 2025-07-11:
pre-chibi migration,

ggurov, 2025-07-11:
[tps_drop_at_398s_SD.rar.txt](https://github.com/user-attachments/files/21187705/tps_drop_at_398s_SD.rar.txt)

it's actually a rar file

LiQuiD04, 2025-07-13:
Tested on post chibi update f407-disco fw from build server on bench. Drop only on TPS and PPS happens when changing trigger simulator. rawRaw values are valid.

[re_250712_042522.rar.txt](https://github.com/user-attachments/files/21205062/re_250712_042522.rar.txt)

LiQuiD04, 2025-07-14:
I just found that TPS/PPS drops were not random. They appear when hitting burn button, when tune crc in log changes, or when turning on/off trigger simulator.

[image]