rusefi/rusefi #8380: sd card logs always truncated mid record

ggurov, 2025-08-01:
**Describe the bug**
sd card logs always show "Entire file not loaded" "end of file on record 9xxx" 

**Hardware**
 - Board: any

**To Reproduce**
Steps to reproduce the behavior:
1. log on sd
2. get log from card
3. open log
4. See error

**Expected behavior**
logs to be truncated at the record's end so such that they could be concatenated in mlv 

**Screenshots**

[image]

dron0gus, 2025-08-03:
@ggurov can you provide few details?
1. actual size of corrupted file
2. way you connecting ECU to PC. Do you power it off, then connect to PC. Or do you connect PC to running ECU?
3. do you observe same issue if switching ECU vs MSD mode on running device with TS buttons?

zed65, 2025-08-03:
All files on SD is 32768kb even if they only recorded 5 seconds.
The files that are "full" eg. you let the ecu start a new file when the first is full, they are OK to open.
Files that only contains let's say 5 sec of run time then ignition off, they get corrupted.

I turned ignition off then connect PC.

dron0gus, 2025-08-03:
@zed65 it is expected behavior in case of turning ECU off.
Current implementation allocates 32Mb when starting new log file. This gives us better write speed.
If ECU is killed with ign switch - file is not properly closed/truncated to actual size and contains some garbage after last write.

zed65, 2025-08-03:
Ok, so how are we supposed to kill/shutoff the engine/ECU?