rusefi/rusefi #8390: migrate complete rusefi_enums.h to proper enums

rusefillc, 2025-08-03:
rusefi_enums.h should be migrated to modern safer enums

FDSoftware, 2025-08-03:
> sounds like perfect task for LLM? open question: need proper setup/toolset
> 
> [@FDSoftware](https://github.com/FDSoftware) any interest to investigate a proper IDE/plugin/toolset integrated setup, migrate some of the enums and teach me how to migrate the rest?

I don't think LLM can determine where the best scope is to put the "using enum" that will be required so as not to have to refactor everything to EnumName::EnumValue, but we can try, now I create a PR migrating a single enum

FDSoftware, 2025-08-06:
Try of migrating with llm:

[image]

> using commits 3cc4cd6 through 8e19add as an example of the changes, I need to replace enum `can_nbc_e`

now with:

> using commits 3cc4cd6 through 8e19add as an example of the changes, I need to replace enum `can_nbc_e`, name should stay the same, also replace all the references to this enum, if using many on a singular scope, is allowed to use `using enum can_nbc_e;`

[image]

this one is good

[image]

wtf:
[image]

eeee, I would give it a 6/10 at least for trying:
[image]

[chat-d4bfe602-73e9-4c74-8116-98ff9af550af.txt](https://github.com/user-attachments/files/21609479/chat-d4bfe602-73e9-4c74-8116-98ff9af550af.txt)

[chat-333f5ee9-d8fc-461a-9f54-ecfcb7b3f98a.txt](https://github.com/user-attachments/files/21609474/chat-333f5ee9-d8fc-461a-9f54-ecfcb7b3f98a.txt)