rusefi/rusefi #8392: any change with engine running causes AFR dip

rusefillc, 2025-08-03:
engine running
change cranking dwell
hit burn

"unsaved changes" stays yellow
engine dips a bit

rusefillc, 2025-08-03:
@ggurov is this something you are aware of?

[2025-08-03_16.35.37.zip](https://github.com/user-attachments/files/21567986/2025-08-03_16.35.37.zip)

ggurov, 2025-08-04:
@rusefillc : yes aware, and somewhat painful. one of the biggest troubles with this is that ETB runs away during the blip. engine revs to like 2-3k after closing a table after editing it.

it's somewhere around where requested burn tears down all of the sensors, unregisters them, and re-registers them anew.

mck1117, 2025-08-04:
> this is that ETB runs away during the blip. engine revs to like 2-3k after closing a table after editing it.

@ggurov something is catastrophically wrong with your throttle tune

ggurov, 2025-08-04:
> > this is that ETB runs away during the blip. engine revs to like 2-3k after closing a table after editing it.
> 
> [@ggurov](https://github.com/ggurov) something is catastrophically wrong with your throttle tune

it's possible, however, there is still a blip in the data when there shouldn't be one.

mck1117, 2025-08-04:
> this isn't cranking. this is while running. 

Your bias table is probably wrong, and also the ETB PID is getting an over-eager reset

ElDominio, 2025-08-04:
> snip

Refer to my first comment

rusefillc, 2025-08-04:
> I think this is because ECU think's it came from Cranking state, since everything related to cranking/afterstart happens after a burn. Idle duty begins from cranking and does transition into normal idle.

please clarify if you see this in the logs or of that's just your expert opinion?

[image]

rusefillc, 2025-08-04:
@dron0gus 
```
2025-08-04_08_03_36_646: EngineState: USB: not ignoring F
2025-08-04_08_03_36_653: EngineState: USB: got Query command
2025-08-04_08_03_36_657: EngineState: TS <- Query signature: rusEFI master.2025.08.03.pnp-e38.2186355567
2025-08-04_08_03_36_661: EngineState: TS <- Get CRC page 0 offset 0 count 63944 result A9515104
2025-08-04_08_03_36_667: EngineState: TS <- Get CRC page 256 offset 0 count 256 result 5885960D
2025-08-04_08_03_36_671: EngineState: TS <- Get CRC page 512 offset 0 count 2048 result 414554BF
2025-08-04_08_03_36_680: EngineState: TS <- Page 512 read chunk offset 0 count 1024
2025-08-04_08_03_36_686: EngineState: TS <- Page 512 read chunk offset 1024 count 1024
2025-08-04_08_03_36_690: EngineState: TS <- Get CRC page 0 offset 0 count 63944 result A9515104
2025-08-04_08_03_36_697: EngineState: TS -> Page 0 write chunk offset 1642 count 1 (output_count=6267)
2025-08-04_08_03_36_701: EngineState: TS -> Burn
2025-08-04_08_03_36_706: EngineState: unregistering VVT Actuator
2025-08-04_08_03_36_712: EngineState: unregistering NONE
2025-08-04_08_03_36_716: EngineState: unregistering NONE
2025-08-04_08_03_36_720: EngineState: unregistering NONE
2025-08-04_08_03_36_730: EngineState: Scheduling write
2025-08-04_08_03_36_736: EngineState: Burned in 4.2ms
2025-08-04_08_03_36_740: EngineState: TS <- Get CRC page 0 offset 0 count 63944 result 0962A9DD
2025-08-04_08_03_36_744: EngineState: USB: got Query command
2025-08-04_08_03_36_748: EngineState: TS <- Query signature: rusEFI master.2025.08.03.pnp-e38.2186355567
```

ggurov, 2025-08-04:
frequency sensors seem to blip at the Tune CRC16 change
[image]

rusefillc, 2025-08-04:
@ElDominio am I getting close to cause of dip?

[image]

ggurov, 2025-08-04:
@rusefillc : it's the tps accumulator for tps AE detection, that gets reset and not populated with default values. so anything is a change from 0 to current TPS.

commit f18fe77d (2025-08-04): unneeded ETB PID reset causes AE and causes AFR dip fix #8392
only:uaefi
closed 2025-08-04 by rusefillc