rusefi/rusefi #8411: Let's try AI generator on specific Subaru

rusefillc, 2025-08-08:
@dron0gus please try https://alphaxpr.com/pages/tuner-generator-lite

@ElDominio FYI

dron0gus, 2025-08-08:
Diff vs my current (bad) tune.
Current tune is mostly defaults.
Left is my, right is generated.
More fuel:
[image]

More realistic VE. My VE was autogenerated by TS + left bottom corner is autotuned a bit (until I killed a AFR sensor). Autogenerated values was crazy. Not sure why. AI at least are close to what I have tuned with autotune.
[image]

Some MAP estimate table. My was zeroed.
[image]

Injection timing is absolutely different. And confusing TPS/TPS fuel correction. Is it multiplier or adder?
[image]

Ignition table. I'm not a tuner, I dont understand difference. What is this area of 38* for? Engine braking?
[image]

Dwell tables. Do I need different base Dwell vs RPM on individual COPs? Battery correction looks similar.
[image]

Base idle position. I was incorrectly  understand idea of this table. AI have fixed my mistake.
[image]

ElDominio, 2025-08-08:
> Diff vs my current (bad) tune. Current tune is mostly defaults. Left is my, right is generated. More fuel: snip
> 
tuner genrator plays very safe especially if engine is not turbo from the factory

> More realistic VE. My VE was autogenerated by TS + left bottom corner is autotuned a bit (until I killed a AFR sensor). Autogenerated values was crazy. Not sure why. AI at least are close to what I have tuned with autotune. snip">
> 
with proper injector data (or at least close-ish, which this can generate!) many VE tables look like the one on the right
> Some MAP estimate table. My was zeroed. snip
> 
MAP estimate is used to do transient fueling instead of using MS adder. MS adder is a crapshoot, MAP Prediciton is a crapshoot at least somewhat grounded in reality
> Injection timing is absolutely different. And confusing TPS/TPS fuel correction. Is it multiplier or adder? snip
> 

[image]

Multiplier. We need less transient correction vs RPM since MAP changes much faster (engine is spinning faster). MAP Prediction is usually good enough to the point that this table doeosn't matter (since MAP will be >= predicted MAP quite quickly)

> Ignition table. I'm not a tuner, I dont understand difference. What is this area of 38* for? Engine braking? snip
> 
38 degrees is for low load cruising, more timing = more time to burn = more efficient

> Dwell tables. Do I need different base Dwell vs RPM on individual COPs? Battery correction looks similar. snip
> 
You have less time to dwell vs RPM, but this is purely theoretical. I base this off the fact that most factory dwell compensations look like this

> Base idle position. I was incorrectly understand idea of this table. AI have fixed my mistake. snip

yay!

closed 2025-09-19 by rusefillc