rusefi/rusefi #8418: Idle RPM Target is not following table

ElDominio, 2025-08-10:
**Describe the bug**
Table shows 600RPM, idleTarget is 905. Changing target table changes nothing. Lua is not changing target RPM (if that is even possible)

**Hardware**
 - Board: Platinum F7 running 20250810
 - Engine info Voodoo

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on engine
2. wait for idle to "come down"
3. idle never "comes down", closed loop is targetting 900 instead of my idle target RPM

**Expected behavior**
idleTarget follows idle target RPM table

**Screenshots**

[image]

**Additional context**
Log and tune

[idleerrormigt350.zip](https://github.com/user-attachments/files/21707066/idleerrormigt350.zip)

ElDominio, 2025-08-10:
[image]

Table is showing me we are using values from 212F, but it is actually using the "97f" row. Lowering around this area change idleTarget to 700

rusefillc, 2025-08-10:
@FDSoftware this might be first freedom units defect?

FDSoftware, 2025-08-10:
> @FDSoftware this might be first freedom units defect?

I think so, I'll check it now

FDSoftware, 2025-08-11:
> Table is showing me we are using values from 212F, but it is actually using the "97f" row. Lowering around this area change idleTarget to 700

[image]

I suspect this bug is because the bins in C have a scale of 2, but when converted to F, they remain at 1.

(Internally, we copy the scale as-is in C and ignore it for F.)

see:
212F / 2 => 106F => closet is 104F cell

rusefillc, 2025-08-11:
@FDSoftware would it help if we https://github.com/rusefi/rusefi/issues/8420 and https://github.com/rusefi/rusefi/commit/05aeb02df57b534e0dd975a6b1e99823c05c3027

rusefillc, 2025-08-11:
@ElDominio could be fixed, would have to migrate project in C

rusefillc, 2025-08-20:
@ElDominio test or close?

ElDominio, 2025-09-04:
Every time i switch to imperial something breaks very badly and i use my Mustang frequently so testing is etremely painful. Someone with more dire need of imperial should be testing

closed 2025-09-04 by rusefillc