rusefi/rusefi #8433: Crank-to-Run idle with AC On does not use A/C compensation

ElDominio, 2025-08-13:
**Describe the bug**
When we transition from Crank to Run, we should use A/C idle offset, since weak cars tend to turn the compressor on here and they die due to lack of oxygen.

**Hardware**
 - Board: paralela
 - Engine info 1NZFE

**To Reproduce**
Steps to reproduce the behavior:
1. Start car with AC button ON
2. look at Idle Duty
3. Notice that no AC compensation is active.
4. Once we get to idling state, a/c offset is suddendly added

**Expected behavior**
A/C idle offset is present at all times EXCEPT coasting/cruising/wot

**Additional context**
[yarispnp.zip](https://github.com/user-attachments/files/21745401/yarispnp.zip)

tune and logs

rusefillc, 2025-08-25:
@ElDominio please help by adding a screenshot of log with relevant fields

ElDominio, 2025-09-16:
https://youtu.be/ZoMirIIb-B8

aaaaaaaaaaaaaaaaaaaaa

rusefillc, 2025-09-16:
@ElDominio was this fixed by https://github.com/rusefi/rusefi/pull/8591?

closed 2025-09-16 by ElDominio