rusefi/rusefi #8434: "setAnitLagTrigger(val)" in Lua

rusefillc, 2025-08-13:
Same as https://github.com/rusefi/rusefi/issues/8400 but for anti lag

[image]

ElDominio, 2025-08-13:
Rolling antilag is actually way more similar to Launch Control than antilag ALS.  

Will open seperate ticket about renaming antilag ALS; 

this feature request is launch control, where RPM is set  in real time at the moment of the button press so ignition is retarded and cut at the rpm the button is pressed at (so when the user is at speed, they can build boost)

rusefillc, 2025-08-13:
@ElDominio do you mean "setRollingIdleTrigger" is a much better name? we still need Lua for whatever this? anything with HW enable should have lua enable?

ElDominio, 2025-08-13:
I never said the word Idle?
Existing setLaunchTrigger can be used to make a rolling launch inside lua

ElDominio, 2025-08-13:
ROTATIONAL idle has no use outside of idle? I am not very well versed on it's uses outside of rally cars and TikTok slop

rusefillc, 2025-08-13:
My typo, yes rotational idle