rusefi/rusefi #8445: MAP is affected by AN2834 3.2.7 "Effect of the analog signal source resistance"

rusefillc, 2025-08-15:
[image]

dron0gus, 2025-08-15:
https://github.com/rusefi/rusefi/issues/976 related

dron0gus, 2025-08-16:
Ok. There are two issues:
1. Internal sampling capacitor current causing noise/needles. Covered in AN2834. Solutions:
  - lower impedance of source cascade - lower values of divider resistors.
  - add **small** caps on ADC inputs. Article recommends something about 200-300pF. Should be calculated in accordance with signal source resistance.
  - ignore or compensate in SW. Simplest solution. Need to check error over all input range and decide if it affects/cares us.
Changing divider and adding cap should be aligned with opamp. We should check DS of used opamp if it capable to work with capacitance / low resistance load.
2. simultaneous digitizing of same input with "slow" and "fast" ADC. Loading same pin with two ADC with random overlap cause significant noise in ADC values:
[image]

dron0gus, 2025-08-29:
After https://github.com/rusefi/rusefi/pull/8517 we have almost constant error over whole input range

[image]