rusefi/rusefi #8447: Torque reduction while using "Clutch Up Switch" trigger is inverted.

ElDominio, 2025-08-17:
**Describe the bug**
When clutch up is RED (meaning clutch is NOT depressed), this is treated as whatever pin that depends on it being trigger.

**Hardware**
 - Board: H154
 - Engine info BKTheta

**To Reproduce**
Steps to reproduce the behavior:
1. set trigger condition for Torque Reduction as Clutch Up switch
2. when the user is not stepping on pedal, Clutch Up is RED, and Torque Reduction trigger is also RED. I need that when specifically Clutch Up is RED, Torque Reduction trigger is WHITE. Clutch Down should work like this (when Clutch Down is RED, Torque Reduction trigger is RED.

**Expected behavior**
 I need that when specifically Clutch Up is RED, Torque Reduction trigger is WHITE. 

**Additional context**
Current workaround: set trigger mode to INVERTED DEFAULT. This produces a bold faced LIE that turns the Clutch Up bulb RED when the clutch is DOWN, making it work correctly

rusefillc, 2025-08-17:
@ElDominio please add some relevant screenshots

ElDominio, 2025-08-17:
[image]

bad behavior, correct setting

[image]

correct behavior, bad setting

rusefillc, 2025-08-17:
@ElDominio 

[image]

rusefillc, 2025-08-17:
@ElDominio 

[image] actually no, it has no name. What is this switch - is it clutch UP or clutch DOWN?