rusefi/rusefi #8515: java OpenBLT frontend is not reliable

rusefillc, 2025-08-28:
happy update but device does not reboot

power cycle does not help - looks like damaged flash process? 

[image]

rusefillc, 2025-08-28:
@esaulenka see https://github.com/rusefi/rusefi/commit/28e7e3885b66bf81d20f828fb7f9a156734f61bd

esaulenka, 2025-09-13:
I've checked it on UAEFI with STM32F407

The root issue is in our `flash_int.cpp`
So, what do we have:
- sector 8 has addresses 0x0808'0000 .. 0x0809'ffff
- sector 9 has addresses 0x080A'0000 .. 0x080B'ffff
- host asking to erase 0x0809'8000 .. 0x080a'2bc8 (for the last firmware chunk)
- `intFlashErase()` successfully erases sector 8, but requested size (43k) is less than sector size (128k). That means that sector 9 remains unerased.

In the PR I propose 2 fixes:
- workaround for host to fix it with the exising bootloader. Use the smaller erase size to be sure that we'll always erase the only sector (at least with the current flash configuration, while application starts at 0x0800'8000)
- fix for the erase function itself

rusefillc, 2025-09-13:
@dron0gus пожалуйста помоги понять корень проблемы

closed 2025-09-26 by rusefillc