rusefi/rusefi #8520: TunerStudio difference report for LTFT table

rusefillc, 2025-08-29:
User asks about TS showing difference report after LTFT progress without TS

TS has a bunch of key words for ECU preference to do we need to leverage some of those?

ggurov, 2025-08-29:
difference report shows up when TS detects the tune in the controller is different from local. 
 i don't think it will come up with the way ltft is stored now

dron0gus, 2025-08-29:
https://github.com/rusefi/rusefi/blob/03fdefc39067ae3312be78ebe9f0e3e5a0aa0e5c/firmware/integration/config_page_2.txt#L3

`noMsqSave,controllerPriority`

```
controllerPriority – Will save the value to the Calibration file and load it from there if not online with a
controller. However, any time a controller is connected, the value from the controller is always silently
taken and replaces any value is in the local store.
controllerPriority Constants can only be changed while connected to the controller. Offline tuning can not
be performed as any change will be unquestionably taken from the controller on connect.
```

dron0gus, 2025-08-29:
Users should not see Diff dialog if LTFT was changed by ECU and this is the **only** diff