rusefi/rusefi #8521: MAP gauge does not show PSI

rusefillc, 2025-08-29:
``
#if USE_METRIC_UNITS
	MapPressure = { MAPValue }
#else
	MapPressure = { MAPValue }
#endif
``

[image]

rusefillc, 2025-08-29:
@FDSoftware gauge baro was not converted as well?

FDSoftware, 2025-08-30:
> [@FDSoftware](https://github.com/FDSoftware) gauge baro was not converted as well?

converted, we also have all of these only in metric:

- AuxT1Gauge
- AuxT2Gauge
- ambientTempGauge
- lowFuelPressureGauge
- fuelTempGauge
- acPressureGauge
- egt1Gauge to egt8Gauge

rusefillc, 2025-08-30:
> converted, we also have all of these only in metric:

@FDSoftware still open issue or covered by same fix? i.e. what do we do about this ticket here?

FDSoftware, 2025-08-30:
I would say close this and add the missing gauges as scope of the main ticket (4788)

closed 2025-08-31 by rusefillc