rusefi/rusefi #8587: tuning detector is working for manual tuning but not for automated tuning

rusefillc, 2025-09-16:


dron0gus, 2025-09-17:
TunnerStudio updates ECU with new VE values every 15 seconds. (at least in my tests)
Should we increase Tunning detection timeout to 20 seconds?
```
$ cat ~/TunerStudioProjects/TunerStudioAppDebug.txt  | grep 'Write Ch'
9/17/25, 11:46:03 AM :Debug: Time:24.403s. Received Instruction: Write Chunk, Page: 1
9/17/25, 11:46:03 AM :Debug: Time:24.725s. Processing Instruction: Write Chunk, Page: 1
9/17/25, 11:46:03 AM :Debug: Time:24.728s. Processing Complete: Write Chunk
9/17/25, 11:46:18 AM :Debug: Time:39.392s. Received Instruction: Write Chunk, Page: 1
9/17/25, 11:46:18 AM :Debug: Time:39.726s. Processing Instruction: Write Chunk, Page: 1
9/17/25, 11:46:18 AM :Debug: Time:39.729s. Processing Complete: Write Chunk
9/17/25, 11:46:33 AM :Debug: Time:54.462s. Received Instruction: Write Chunk, Page: 1
9/17/25, 11:46:33 AM :Debug: Time:54.727s. Processing Instruction: Write Chunk, Page: 1
9/17/25, 11:46:33 AM :Debug: Time:54.729s. Processing Complete: Write Chunk
9/17/25, 11:46:48 AM :Debug: Time:69.461s. Received Instruction: Write Chunk, Page: 1
9/17/25, 11:46:48 AM :Debug: Time:69.729s. Processing Instruction: Write Chunk, Page: 1
9/17/25, 11:46:48 AM :Debug: Time:69.731s. Processing Complete: Write Chunk
```

rusefillc, 2025-09-17:
@ElDominio can you explain why do you insist that STFT is enabled while you are touching VE?

closed 2025-09-19 by rusefillc